tscircuit/docs #126: Create a guide on running SPICE simulations in tscircuit

seveibar, 2025-09-05:


zalo, 2025-09-06:
I wonder if this should be a tab alongside Scematic/Layout/3D 🧐

(If the simulation is wholly defined in code)

seveibar, 2025-09-18:
@zalo i kind of agree with having a separate tab for SPICE, we just don't know what that should look like (lmk if you're interested in brainstorming some possible designs) current SPICE is available in the schematic viewer via "simulate icon", so it's basically hidden

@ShiboSoftwareDev (also @zalo if you're interested) we should mock up some designs for what a SPICE tab could look like, i think we need some design references

I know personally, i'd like to see the schematic at the same time as i see waveforms, currently the waveforms obscure the schematic. But we should definitely find a couple design references before doing too much experimentation

closed 2025-12-02 by github-actions[bot]